RISE-Maritime/keelson #7: [`lidar`] Add support for IMU packets in ouster connector

freol35241, 2023-11-13:
Currently, the connector (https://github.com/MO-RISE/keelson/blob/main/keelson-interface-lidar/bin/ouster) only make use of the LidarPackets and inserts these into a PointCloud. 

We also want the raw IMU readings from the sensor, see https://static.ouster.dev/sdk-docs/python/examples/udp-packets.html#ex-packets for details.

Requries #6 to be solved first/in parallell.

RISEMaxF, 2023-12-21:
The way I'm approaching this is to use our new common KeelsonScans that inherits from Ouster Scans, and then create an additional topic for IMU data - effectively resulting in a very similar setup to #5 . Is this the approach you would choose as well or do you prefer another solution?

masslab-user, 2024-09-11:
Lidar connector moved to https://github.com/RISE-Maritime/keelson-connector-lidar-os

closed 2024-09-11 by masslab-user